Intelligent-Quads/iq_tutorials #1: Speed

aircraftpilot5, 2020-01-16:
Firstly, thank you for your really good documentation. I want to ask a question about the speed of the drone. Which speed the drone fly in a square? The default speed seems to be 500m/s. Where do you decrease this speed for flying in a square or obstacle avoidance part? Could you please clarify this?

ericjohnson97, 2020-01-18:
I have been using Ardupilot's waypoint nav speed parameter(`WPNAV_SPEED`) to control the speed of my missions. You can also adjust the aggressiveness of your drone by using the `WPNAV_ACCEL` parameter. I have a few of the common parameters documented here https://github.com/Intelligent-Quads/iq_tutorials/blob/master/ardu_params_and_commands.md

Best!

aircraftpilot5, 2020-01-20:
Thank you very much...

closed 2020-01-20 by aircraftpilot5