Intelligent-Quads/iq_tutorials #7: Drone not appearing in gazebo "runway world"

lkw303, 2020-06-23:
Hi i faced this issue where the world launches but the drone is no where to be seen. I am still seeing the topics published and the "iris" object listed on the left in the gazebo GUI but selecting it does not highlight anything in the gazebo world .After running the software in the loop command ```~./sitl.sh``` It is unabble to connect to any drones and shows ```MAVROS``` instead of  ```STABILIZE```

this is what shows in my terminal after launching the runway.launch file
```libcurl: (6) Could not resolve host: api.ignitionrobotics.org
[ INFO] [1592934696.542627127]: Camera Plugin: The 'robotNamespace' param did not exit
[ INFO] [1592934696.544752287, 0.061000000]: Camera Plugin (ns = )  <tf_prefix_>, set to "
```

this shows up after running the sitl.sh file
```
Waiting for heartbeat from tcp:127.0.0.1:5760
```

moreover the world has issues launching with my internet on and does fine after turning it off. Not sure if that is contributing to this issue. (I have already removed modemmanager.)

any help or advice would be appreciated thanks!

lkw303, 2020-06-23:
solved with this thread https://github.com/PX4/avoidance/issues/295. Turns out I didnt have models loaded into gazebo. Not sure why the drone appears in the previous tutorial but not in this one though I think maybe because the different launch files have different paths set to different models?

closed 2020-06-23 by lkw303

aabs7, 2020-07-29:
And if all of that isn't clear.
Do this:
Edit ```.bashrc``` in home directory.
add line ```export GAZEBO_MODEL_PATH=~/catkin_ws/src/iq_sim/models```
It will work.