Intelligent-Quads/iq_tutorials #8: SIM_VEHICLE: Killing tasks  |  RuntimeError: generator raised StopIteration

MuhammadBilal1, 2020-07-01:
I followed your complete tutorial a few weeks ago and this all worked without any trouble.
But after sometime I resumed work on this again yesterday, and now I am getting the below error while running the following command.

`sim_vehicle.py -w`

**Below is the terminal output:**
myuser@ubuntu:~/ardupilot/ArduCopter$ sim_vehicle.py -w
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/myuser/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/myuser/ardupilot/modules/waf/waflib/Node.py", line 590, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/myuser/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/myuser/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/myuser/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/myuser/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/myuser/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/myuser/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/myuser/ardupilot/wscript", line 52, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/myuser/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/myuser/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/myuser/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/myuser/ardupilot/modules/waf/waflib/Node.py", line 683, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/myuser/ardupilot/modules/waf/waflib/Node.py", line 683, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
**RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks**

### **My OS specifications:**

[image]

cmjagtap, 2020-07-02:
Hi @MuhammadBilal1 did you solve problem??

MuhammadBilal1, 2020-07-02:
> Hi @MuhammadBilal1 did you solve problem??

**No**

arthursn138, 2020-07-13:
Same issue here.
Any luck @MuhammadBilal1 and @cmjagtap ? 
I'm guessing you guys also had the python/python3 issue, right?
I solved it with `sudo update-alternatives --install /usr/bin/python python /usr/bin/python3 10`

@ericjohnson97 could you shed some light?

cmjagtap, 2020-07-14:
@arthursn138  Yes that's python issue. Thanks man.

arthursn138, 2020-07-14:
Did that solve it @cmjagtap ?

Because I'm still getting the errors:

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/arthur/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/arthur/ardupilot/modules/waf/waflib/Node.py", line 591, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/arthur/ardupilot/modules/waf/waflib/Scripting.py", line 166, in waf_entry_point
    run_commands()
  File "/home/arthur/ardupilot/modules/waf/waflib/Scripting.py", line 263, in run_commands
    parse_options()
  File "/home/arthur/ardupilot/modules/waf/waflib/Scripting.py", line 215, in parse_options
    Context.create_context('options').execute()
  File "/home/arthur/ardupilot/modules/waf/waflib/Options.py", line 272, in execute
    super(OptionsContext, self).execute()
  File "/home/arthur/ardupilot/modules/waf/waflib/Context.py", line 206, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/arthur/ardupilot/modules/waf/waflib/Context.py", line 288, in recurse
    user_function(self)
  File "/home/arthur/ardupilot/wscript", line 52, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/arthur/ardupilot/modules/waf/waflib/Context.py", line 198, in load
    fun(self)
  File "/home/arthur/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/arthur/ardupilot/modules/waf/waflib/Context.py", line 610, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/arthur/ardupilot/modules/waf/waflib/Node.py", line 684, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/arthur/ardupilot/modules/waf/waflib/Node.py", line 684, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks

arthursn138, 2020-07-16:
Another info I could extract from the output was the lack of the waf directory. [This issue](https://github.com/ArduPilot/ardupilot/issues/5881) helped me progress, it's basically an update issue on some submodules. Some people resolved with `git submodule update --init` under `cd ~/ardupilot/modules/waf`, but I still have some issues.

Let me know if that worked for you guys of if you had any progress.

ericjohnson97, 2020-08-16:
Sorry guys. just noticed this issue. I'm going to be better about responding to these going forward.  like @arthursn138 alluded to build issues can be cause by not having the correct branch and submodules checked out. What works best for me when switching branches is to run the following
```
git checkout <branch>
git submodule update --init --recursive -f
```
use  --recursive to get the submodules of the submodules. and use -f to make sure that git actually tries to ckeckout and pull all the correct submodules.

closed 2020-08-30 by ericjohnson97

rajpal007, 2022-01-24:
$ ../Tools/autotest/sim_vehicle.py --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/aditi/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 579, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/aditi/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/aditi/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/aditi/ardupilot/wscript", line 49, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/aditi/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/aditi/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 672, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/aditi/ardupilot/modules/waf/waflib/Node.py", line 672, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: a bytes-like object is required, not 'str'

PLEASE HELP I UNINSTALLED PYTHON 3 . Coz ppl said its python2/3 issue

iPAS, 2023-01-31:
> @arthursn138 Yes that's python issue. Thanks man.

Python2.7 is ok.

harouneerachide, 2024-06-28:
help me please 
sim_vehicle.py -w
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/rachideu/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 590, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
    run_commands()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
    parse_options()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
    Context.create_context('options').execute()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
    super(OptionsContext, self).execute()
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
    user_function(self)
  File "/home/rachideu/ardupilot/wscript", line 52, in options
    opt.load('compiler_cxx compiler_c waf_unit_test python')
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 197, in load
    fun(self)
  File "/home/rachideu/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
    opt.load_special_tools('cxx_*.py')
  File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
    lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in ant_glob
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
  File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in <listcomp>
    ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks