Intelligent-Quads/iq_tutorials #9: error: ‘set_mode’ was not declared in this scope

romi-seecs, 2020-07-11:
Hi!
I was following the tutorial "Drone Swarms with Ardupilot+MAVROS" and as suggested used set_mode() function to set mode to guided in multi_drone.cpp and got this error:

```
/home/ans/catkin_ws/src/iq_gnc/src/multi_square.cpp:20:2: error: ‘set_mode’ was not declared in this scope  set_mode("GUIDED");
  ^~~~~~~~
/home/ans/catkin_ws/src/iq_gnc/src/multi_square.cpp:20:2: note: suggested alternative: ‘gnc_node’
  `set_mode("GUIDED");
```
Need some help @ericjohnson97  :/

ericjohnson97, 2020-07-26:
oh haha. That tutorial was still in development. Completed the tutorial today. git pull for the latest iq_gnc and iq_sim. should be fixed now,

romi-seecs, 2020-07-27:
Yup, Got fixed :+1: :)

closed 2020-07-27 by romi-seecs