Intelligent-Quads/iq_tutorials #13: Change initial psi of drone

taminhtoan2601, 2020-11-02:
Hello, 

Could I can change the initial psi ? Because I realize that initial value of local_offset_g (psi) is about 90 degree, so the local coordinate frame is rotated about 90 degree, because I want the local coordinate system to coincide with the ENU coordinate system in gazebo (local_offset_g = 0). 
Thank you very much for responding me.

ericjohnson97, 2020-11-03:
the way to change the orientation of the drone in global coordinates is through the sitl. The location and orientation is determined by this file.  

https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/locations.txt

I believe the default is `CMAC`. You can create your location by following the format and adding the lat,lot,alt,haeding of the start location.  to use your location use `-L <name of location>` when running `sim_vehicle.py` 

If you want to change the start orientation of the drone relative to the drone you can change it in the world file.
https://github.com/Intelligent-Quads/iq_sim/blob/master/worlds/runway.world#L85

taminhtoan2601, 2020-11-03:
Hello, 

Thank you for your answer, I modified values of CMAC in file location.txt (for example, CMAC=-35.363261,149.165230,584,0 with heading = 0 ) but the initial value of psi is still about 85-90 degree... How can I modify this value so that it's about 0 degree? Thank you very much!

ericjohnson97, 2020-11-03:
First thing I would do is connect qgroundcontrol and and make sure the compass heading you expect to see is being shown on the ground station. If is not, it is likely an issue with mavros. The way mavros formulates the local position is it takes the initial ekf position and orientation from the fcu when mavros is run and stores it. If you run mavros before the ekf has a 3d fix or rerun the after rebooting ardupilot the plugin can show incorrect position and orientation.

closed 2020-11-25 by ericjohnson97