Intelligent-Quads/iq_tutorials #14: "roslaunch  iq_sim apm.launch" fails on Ubuntu 20.04 + ros-noetic + gazebo11 + catkin_make_isolated:

AJRepo, 2020-11-25:
I've been re-doing the setup of a new, virtual (qemu/KVM) machine a few times to try to get a script that creates a good, stable image with ros-noetic, gazebo11 and Ubuntu 20.

The step "[Install ROS and Setup Catkin](https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/installing_ros_20_04.md)"  seemed to go fine with no errors although with the package install of catkin there is no command "catkin"  but instead only "catkin_make_isolated" and "catkin_make"

The [next step](https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/ros_intro.md)  gets into an infinite restart loop with roslaunch iq_sim apm.launch

Example:

```
$ roslaunch iq_sim apm.launch
... logging to /REDACTED/.ros/log/8053100a-2f2d-11eb-9030-43a91962291d/roslaunch-gaz1-2230.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gaz1:36233/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://127.0.0.1:1...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
[... 245 lines cut ...]
[ INFO] [1606316413.592512702]: udp0: Remote address: 127.0.0.1:14555
[ INFO] [1606316413.592789264]: GCS bridge disabled
```

(note there is a udp0 remote address line)

or

`<arg name="fcu_url" default="udp://127.0.0.1:14551@:14555" />` 

which gives

```
[ INFO] [1606316558.272979009]: FCU URL: udp://127.0.0.1:14551@:14555
[ INFO] [1606316558.275885652]: udp0: Bind address: 127.0.0.1:14551
[ INFO] [1606316558.276202900]: GCS bridge disabled
```

(note there is again no remote address line). 

I suspect the core issue is probably the newer code requires a ":" in the remote address field for specifying the port. 

In all 3 cases, the GUI for gazebo does not start. 

Since I'm new to gazebo/ros and also not working with Gazebo9 or ros-melodic any more I didn't want to submit any of those changes as a patch as I don't know enough to know if it is is backwards compatible or if perhaps there is something else I've done wrong.

zdanek, 2021-02-22:
I have the same problem.
It hangs forever at
[ INFO] [1614020735.714864508]: MAVROS started. MY ID 1.240, TARGET ID 1.1

Same is when Ardupilot SITL is run and when it's not. It seems like mavros is somehow waiting to connect to the SITL but it never happens.

Any clues?

aaryanmurgunde, 2022-02-14:
@AJRepo This issue is dude to some error with the mavros try reinstalling it by following the [guidelines](https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation)

wildangunawan, 2022-02-16:
I have this issue as well. Thank to @AJRepo, I managed to launch MAVROS successfully. I'm following how PX4 launches MAVROS (only the format), you can look at [here](https://github.com/PX4/PX4-Autopilot/blob/master/launch/mavros_posix_sitl.launch#L25).

@AJRepo did say that apm.launch didn't start Gazebo which is correct. This is due to the launch file itself doesn't start the Gazebo (refer to how runway.launch does start Gazebo).

In case we want MAVROS and Gazebo to start, we can use this:

```xml
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- Experimental launch script for APM based Drone -->

	<!-- FCU URL format stolen from PX4's launch file -->
	<!-- https://github.com/PX4/PX4-Autopilot/blob/master/launch/mavros_posix_sitl.launch#L25 -->
	<arg name="fcu_url" default="udp://:14551@localhost:14555" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="respawn_mavros" default="true"/>
	<arg name="mavros_ns" default="/"/>
	<arg name="config_yaml" default="$(find mavros)/launch/apm_config.yaml" />

	<include file="$(find iq_sim)/launch/mavros_node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/apm_pluginlists.yaml" />
		<arg name="config_yaml" value="$(arg config_yaml)" />
		<arg name="mavros_ns" value="$(arg mavros_ns)"/>
		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="respawn_mavros" value="$(arg respawn_mavros)"/>
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
	</include>
	
	<!-- Code to launch Gazebo stolen from runway.launch file -->
	<!-- We resume the logic in empty_world.launch, changing only the name of the world to be launched -->
	<arg name="gui" default="true"/>
		<include file="$(find gazebo_ros)/launch/empty_world.launch">
		<arg name="world_name" value="$(find iq_sim)/worlds/runway.world"/>
		<arg name="gui" value="$(arg gui)"/>
		<!-- more default parameters can be changed here -->
	</include>
</launch>
```

You can either edit directly at apm.launch or create your own launch file.

ARidwanW, 2023-06-07:
That is also a problem for me. I tried the solutions you suggested, but the problem persists. When I try to change fcu to match @wildangunawan 's solution, I get something like @zdanek.

After that, I discovered this video: [Ardupilot Mavros connexion](https://www.youtube.com/watch?v=LRo2G3z6s5w)

utilizing the fcu:

`<arg name="fcu_url" default="tcp://localhost:5762" />`

It worked for me, but when I ran square.cpp, the drone did not form a square but instead immediately rotated 180 degrees or faced back after takeoff and advanced 5 meters.

ARidwanW, 2023-06-07:
then i try to debug the MAVROS node:

`gdb -ex run --args /home/arw/catkin_ws/devel/lib/mavros/mavros_node __name:=mavros` then type `bt`

then i got this:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7405859 in __GI_abort () at abort.c:79
#2  0x00007ffff768e911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff769a38c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff769a3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff769a6a9 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff769173f in std::__throw_system_error(int) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff76c7060 in std::thread::join() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff73698eb in mavconn::MAVConnUDP::close() () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#9  0x00007ffff736a744 in mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}::operator()(boost::system::error_code, unsigned long) const () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#10 0x00007ffff736d725 in boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>::operator()() () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#11 0x00007ffff736d404 in void boost::asio::asio_handler_invoke<boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long> >(boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&, ...) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#12 0x00007ffff736d038 in void boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>, {lambda(boost::system::error_code, unsigned long)#1}>(boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&, {lambda(boost::system::error_code, unsigned long)#1}&) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#13 0x00007ffff736cda1 in void boost::asio::detail::asio_handler_invoke<boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>, {lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>(boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&, boost::asio::detail::binder2<{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>*) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
--Type <RET> for more, q to quit, c to continue without paging--c
#14 0x00007ffff736ca45 in void boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>, mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}>(boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&, mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}&) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#15 0x00007ffff736c7d8 in void boost::asio::detail::io_object_executor<boost::asio::executor>::dispatch<boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>, std::allocator<void> >(boost::asio::detail::binder2<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&&, std::allocator<void> const&) const () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#16 0x00007ffff736c5c3 in void boost::asio::detail::handler_work<mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::asio::detail::io_object_executor<boost::asio::executor>, boost::asio::executor>::complete<boost::asio::detail::binder2<{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long> >(boost::asio::detail::binder2<{lambda(boost::system::error_code, unsigned long)#1}, boost::system::error_code, unsigned long>&, {lambda(boost::system::error_code, unsigned long)#1}&) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#17 0x00007ffff736c019 in boost::asio::detail::reactive_socket_sendto_op<boost::asio::mutable_buffers_1, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, mavconn::MAVConnUDP::do_sendto(bool)::{lambda(boost::system::error_code, unsigned long)#1}, boost::asio::detail::io_object_executor<boost::asio::executor> >::do_complete(void*, boost::asio::detail::scheduler_operation*, boost::system::error_code const&, unsigned long) () from /home/arw/catkin_ws/devel/lib/libmavconn.so
[... cut ...]
#27 0x00007ffff736db00 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<mavconn::MAVConnUDP::connect(std::function<void (mavlink::__mavlink_message const*, mavconn::Framing)> const&, std::function<void ()> const&)::{lambda()#1}> > >::_M_run() () from /home/arw/catkin_ws/devel/lib/libmavconn.so
#28 0x00007ffff76c6de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#29 0x00007ffff78cb609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#30 0x00007ffff7502133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

ARidwanW, 2023-06-20:
Hello guys, after two weeks I found a solution to this problem! at least for me.

## environments:
- Ubuntu 20.04 WSL 2
- ROS Noetic
- Copter-4.4
- --out IP 127.0.0.1 (once resolved)

## startsitl.sh
So when you run `~/startsitl.sh` you will run `sim_vehicle.py` where as `sim_vehicle.py` will run `mavproxy.py`.

from there (`~/startsitl.sh`) you can see if `mavproxy.py --out` is running ip `127.0.0.1`. 

[image]

## sim_vehicle.py
if it doesn't run ip `127.0.0.1`, then you can change the `sim_vehicle.py` file (use: `whereis sim_vehicle.py` for the path) by commenting the following line:

```
for i in instances:
        if not opts.no_extra_ports:
            ports = [p + 10 * i for p in [14550, 14551]]
            for port in ports:
                if under_vagrant():
                    # We're running inside of a vagrant guest; forward our
                    # mavlink out to the containing host OS
                    cmd.extend(["--out", "10.0.2.2:" + str(port)])

                # elif wsl2_host_ip_str:
                #     # We're running WSL2; forward our
                #     # mavlink out to the containing host Windows OS
                #     cmd.extend(["--out", str(wsl2_host_ip_str) + ":" + str(port)])

                else:
                    cmd.extend(["--out", "127.0.0.1:" + str(port)])
```
then save the file and rerun ~/startsitl.sh and you don't need to change the ip on apm.launch. then try again
`roslaunch iq_sim apm.launch`

I hope this can help! :)

prodrone-liu, 2024-09-26:
I was also running into the same problem. Although the `<arg name="fcu_url" default="udp://127.0.0.1:14551@:14555" />` did prevent me from looping the error code, I could not rostopic echo the telemetry data like what [tutorial](https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/ros_intro.md#using-mavros-to-get-telemetry-data-from-the-fcu) did. 

By referring the @ARidwanW 's solution, I switched to `<arg name="fcu_url" default="udp://:14550@localhost:14555" />` and everything works. Port 14450 is referenced according to the output of `sim_vehicle.py -v ArduCopter -f gazebo-iris --console`:

```shell
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--console"
``` 

Once apm.launch is successfully launched, the FCU connection status can be seen by using command `rostopic echo -n1 /diagnostics`:
```shell
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14550@localhost:14555"
    values: 
      - 
        key: "Received packets:"
        value: "1016"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "50"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "33148"
      - 
        key: "Tx total bytes:"
        value: "2628"
      - 
        key: "Rx speed:"
        value: "5997.000000"
      - 
        key: "Tx speed:"
        value: "500.000000"
  - 
```