Intelligent-Quads/iq_tutorials #15: Unable to find the drone model (drone_with_camera)

abhiramdapke, 2020-12-04:
Hello guys,
I followed all the steps to run the gazebo model, but when I am running the command "roslaunch iq_sim runway.launch", it is not finding the uri model - drone_with_camera. I am able to see the empty world without the arducopter. I exported all the drone models into .bashrc while following the steps. Could you guide me on how to load that model in the gazebo?

Note: The sitl gazebo simulation without ROS is running perfectly.

Thank you.

gazebo Version 9
ROS Melodic
Ubuntu 18.04
Mavros 1.5.0
Ardupilot firmware

Error log when running "roslaunch iq_sim runway.launch"

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://abhiram:39931/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [5270]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e336d3c8-3658-11eb-93e3-0800273cbdf0
process[rosout-1]: started with pid [5281]
started core service [/rosout]
process[gazebo-2]: started with pid [5285]
process[gazebo_gui-3]: started with pid [5293]
[ INFO] [1607104097.116456502]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1607104097.117582554]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1607104097.145564613]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1607104097.147727900]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
Error Code 11 Msg: Unable to find uri[model://drone_with_camera]
[ INFO] [1607104120.953956962, 0.387000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1607104120.966385469, 1.018000000]: Physics dynamic reconfigure ready.
[ WARN] [1607104146.096593818, 1566.171000000]: Shutdown request received.
[ WARN] [1607104146.096748501, 1566.171000000]: Reason given for shutdown: [[/gazebo_gui] Reason: new node registered with same name]
^C[gazebo_gui-3] killing on exit
[gazebo-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

ericjohnson97, 2020-12-09:
I apologize I recently made an update to iq_sim which removed iq_sim's dependency on the ardupilot plugin models. Because of this you have to add the following to the bashrc 

```
GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:$HOME/catkin_ws/src/iq_sim/models
```   

this tells gazebo to look in that folder for models when loading. 

the instructions have been updated with this commit b6f7c65884ddbae6d52271a6ac244ee664ee0708

abhiramdapke, 2020-12-09:
Thanks a lot for the explanation, Eric. That is exactly to the point. Yesterday, I figured this out. You have a great day!

abhiramdapke, 2020-12-13:
@ericjohnson97 I have an important question to ask. I did all your tutorials and have setup the environment and ran a simulation of arducopter on that. It is working properly.
Now, according to the tutorials, if I run these commands in separate terminals:

roslaunch iq_sim runway.launch
./startsitl.h
roslaunch iq_sim apm.launch
rosrun iq_gnc square
It is running the square.cpp file on the simulated drone when I switch to mode guided in MAVProxy terminal (which is of arducopter).
So, what I want to do is, I want to send these commands to a real drone that has an offboard computer and uses ardupilot and a pixhawk.
I understand that I need the fcu url and baud rate set for that (while writing the roslaunch mavros/iq_sim apm.launch command) which I already have but the question is how do I type the command ./startsitl.sh because that starts the arducopter mavproxy terminal.
My aim is to have the Mavproxy terminal so that I can type the command ''mode guided'' and then pass the script to it but ./startsitl.sh starts the Mavproxy terminal for the arducopter and not the real drone. So how can I solve this problem when trying to fly a real drone?

Also, I don't have a .sdf file of the drone with me.

It would be really helpful if you could provide some solution to this. Thanks a ton.

ericjohnson97, 2020-12-15:
I want to do tutorials on real drones in the future, but I have been being cheap and there aren't any frames that I like. might do something custom, but we will see.  

You should be able to do the same auto takeoff and guided mode waypoints via QGC, And for the mavros auto missions you can simply change the mode to guided via QGC. That would be the easiest. 

If you really want to use mavproxy, you can probably just launch as shown in the link bellow. 
https://ardupilot.org/mavproxy/docs/getting_started/quickstart.html

You can also ask me questions in my discord. I prioritize questions there first https://discord.gg/xZjXaAf

abhiramdapke, 2020-12-16:
@ericjohnson97 Thanks a lot, Eric. I will check out the links you sent me. 

Also, it would be interesting to watch and learn from the videos with a real drone. I must say your tutorials are awesome. They have helped me big time.

I had a couple of questions regarding the same. I went through all the tutorials and have set up my environment properly, however, while I am running the simulation with square.cpp, the drone is crashing sometimes while reaching the second or third waypoint. Sometimes, it crashes abruptly without even reaching the waypoint. Do you know why this might be happening? The drone crashes abruptly at any point in time. I performed at least 10 simulations with this.

The error I am getting in the console is APM: GPS glitch in red and keep getting yaw re-aligned in yellow.

I wanted to run the scripts on a real drone after I get the simulation correctly.
For doing it on the real drone, I need to run only these two commands in two separate terminals, is that correct?
1. roslaunch iq_sim apm.launch
2. rosrun iq_gnc square

I am a bit concerned about running the script on a real drone if there are crashes during the simulation. Could you shed some light on why this might be happenning? That would be really helpful.

The next time, I'll post questions on discord.
Thank you.

abhiramdapke, 2020-12-16:
@ericjohnson97 Also, would it be safe to run these scripts on a real drone?

closed 2021-02-18 by abhiramdapke