Intelligent-Quads/iq_tutorials #16: Error when launching darknet_ros

ppaulBismuth, 2020-12-10:
Hello,
Following your awesome tutorials, I am stuck on the 11th https://github.com/Intelligent-Quads/iq_tutorials/blob/master/docs/intro_to_yolo.md

After installation of `darknet_ros` and modification of the topic `/webcam/image_raw` I can successfully launch both previous command lines `roslaunch iq_sim hills.launch` and `./startsitl.sh`. But I am unable to run the third command line 'roslaunch darknet_ros darknet_ros.launch'... Here is what I get

`... logging to /home/user/.ros/log/01bad07c-3b06-11eb-b028-283926a1d9bb/roslaunch-user-3252.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user:42447/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /home/user/cat...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /webcam/image_raw
 * /darknet_ros/weights_path: /home/user/cat...
 * /darknet_ros/yolo_model/config_file/name: yolov2-tiny.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov2-tiny.weights
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

ROS_MASTER_URI=http://localhost:11311

process[darknet_ros-1]: started with pid [3271]
[ INFO] [1607618331.065877689]: [YoloObjectDetector] Node started.
[ INFO] [1607618331.069489063]: [YoloObjectDetector] Xserver is running.
[ INFO] [1607618331.070552466]: [YoloObjectDetector] init().
YOLO V3
layer     filters    size              input                output
    0 conv     16  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  16  0.150 BFLOPs
    1 max          2 x 2 / 2   416 x 416 x  16   ->   208 x 208 x  16
    2 conv     32  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  32  0.399 BFLOPs
    3 max          2 x 2 / 2   208 x 208 x  32   ->   104 x 104 x  32
    4 conv     64  3 x 3 / 1   104 x 104 x  32   ->   104 x 104 x  64  0.399 BFLOPs
    5 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    6 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128  0.399 BFLOPs
    7 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    8 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256  0.399 BFLOPs
    9 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
   10 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512  0.399 BFLOPs
   11 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024  1.595 BFLOPs
   13 conv    512  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x 512  1.595 BFLOPs
   14 conv    425  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 425  0.074 BFLOPs
   15 detection
mask_scale: Using default '1.000000'
Loading weights from /home/user/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov2-tiny.weights...Done!
Waiting for image.
CUDA Error: invalid device symbol
CUDA Error: invalid device symbol: Success
[darknet_ros-1] process has died [pid 3271, exit code 255, cmd /home/... logging to /home/user/.ros/log/01bad07c-3b06-11eb-b028-283926a1d9bb/roslaunch-user-3252.log

all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/camera/rgb/image_raw __name:=darknet_ros __log:=/home/user/.ros/log/01bad07c-3b06-11eb-b028-283926a1d9bb/darknet_ros-1.log].
log file: /home/user/.ros/log/01bad07c-3b06-11eb-b028-283926a1d9bb/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

I am running it on Ubuntu 18.04 with a GTX1060. I have tried different drivers of the GPU, I have reinstalled 'daknet_ros' building it with and without gcc6. The log file 'log file: /home/user/.ros/log/01bad07c-3b06-11eb-b028-283926a1d9bb/darknet_ros-1*.log' is empty.

Do you have any idea of what can be the problem?

JSpat49, 2020-12-19:
Hi @ppaulBismuth 

Were you able to figure this out? I too am having the same issue.

Sorry I couldn't be of assistance.

ppaulBismuth, 2020-12-30:
Hi @JSpat49 , unfortunately the issue is still open for me. I first thought I was running on the wrong GPU of my computer, so I forced my nvidia GPU tu run, but the error still remains... Don't know what to do...

JSpat49, 2020-12-31:
Hi @ppaulBismuth , sorry to hear it's still an issue for you, same goes for me. I am relatively new to all of this, and am actually following Intelligent-Quads videos using Ubuntu 18.04 in a virtual box, so I think that may be the problem for me. From what I've read, I'm not entirely sure my host computer's NVIDIA GPU is going to be able to access my virtual machine. I assume you and probably everyone else aren't doing the virtual box method?