Intelligent-Quads/iq_tutorials #25: Unable to Simulate APMrover2

Prasanna-icefire, 2021-05-24:
I tried doing a git check with different versions of APMrovers present.
I then altered my startsitl.sh file and replaced ArduCopter with APMRover2.... The process began and after 30 percent or so it fails and exits.

Build failed

 -> task in 'objs/AP_AHRS' failed (exit status 1): 
	{task 140114098061088: cxx AP_AHRS_NavEKF.cpp -> AP_AHRS_NavEKF.cpp.0.o}
 (run with -v to display more information)
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks

ericjohnson97, 2021-06-02:
I don't have any example rover stuff atm, but you should look at this repo 

https://github.com/SwiftGust/ardupilot_gazebo

I believe the command you need is 
```
sim_vehicle.py -v APMrover2 -f gazebo-rover  -m --mav10 --map --console -I1
```

Prasanna-icefire, 2021-06-02:
Haha yup,
I figured this part out and used it up.
Your tutorials are amazing!Thank you!

ericjohnson97, 2021-06-02:
Glad you figured it out! Best of luck with your project!

closed 2021-06-02 by ericjohnson97