Intelligent-Quads/iq_tutorials #27: Unable to build with Ubuntu 20.04

konnoeric, 2021-06-09:
I followed your instruction and did the same command  but I could not have it done. What is the problem and how to fix this?

[image]

O2KAN, 2021-09-06:
Hi, Have you found a solution?

notshayer, 2021-09-07:
@konnoeric in order to use catkin, one must have a valid installation of ROS, as well as the catkin workspace setup.
Follow the guide [here](https://stackoverflow.com/questions/41234957/catkin-command-not-found)

O2KAN, 2021-09-07:
I think our problem is installing  darknet and building it. @konnoeric got an error at 'git checkout opencv4' like me. Then he continued with the error. Probably, because that he couldn't build it.

konnoeric, 2021-09-07:
@notshayer My installation of ROS is okay because I can properly use another one's program. I think there is a problem with this repository.

emsC-137, 2022-07-15:
after doing the command `git clone https://github.com/kunaltyagi/darknet_ros.git`, cd in to `darknet_ros`, then you'll be able to git check out. At least that's what work for me

shrinivas96, 2022-09-25:
This error just measn you do not have catkin-tools installed. The basic way to install it is `sudo apt-get install python3-catkin-tools`.. Though, if you want to know more (or for other platforms) go to the [catkin tools website](python3-catkin-tools).