Intelligent-Quads/iq_tutorials #30: Submodule update fail

AAK350, 2021-12-06:
As I follow the installation for Ubunto 20.04 , I get the below messages after i run (  $ git submodule update --init --recursive ) 

Cloning into '/home/aak/ardupilot/modules/libcanard'...
fatal: read error: Connection reset by peer
fatal: clone of 'git://github.com/ArduPilot/libcanard.git' into submodule path '/home/aak/ardupilot/modules/libcanard' failed
Failed to clone 'modules/libcanard'. Retry scheduled
Cloning into '/home/aak/ardupilot/modules/libcanard'...
fatal: read error: Connection reset by peer
fatal: clone of 'git://github.com/ArduPilot/libcanard.git' into submodule path '/home/aak/ardupilot/modules/libcanard' failed
Failed to clone 'modules/libcanard' a second time, aborting

aircraftpilot5, 2021-12-10:
The same problem has occurred on my system. Did you fix it?

mertgunduc, 2021-12-10:
I am having the same problem but with different submodule. 

Submodule path 'modules/mavlink/pymavlink': checked out '57a6ab0e64468f214c1bfc8ed36fe0e2a900b2e7'
From https://github.com/DroneCAN/libuavcan
 * branch              3ef4b88d96466bbcf886f74d0ae169a4c8c1bdb0 -> FETCH_HEAD
Submodule path 'modules/uavcan': checked out '3ef4b88d96466bbcf886f74d0ae169a4c8c1bdb0'
fatal: remote error: upload-pack: not our ref 192295c4f9b67f4a20b0eabf74757b6597415f2b
fatal: the remote end hung up unexpectedly
Fetched in submodule path 'modules/uavcan/dsdl', but it did not contain 192295c4f9b67f4a20b0eabf74757b6597415f2b. Direct fetching of that commit failed.
Submodule path 'modules/waf': checked out '51f1529bd20759642e9fd0a8db648da9ba380b18'
Failed to recurse into submodule path 'modules/uavcan'

I guess, https://github.com/UAVCAN/dsdl this module doesnt exist anymore . How can I update my  submodule reference to the new URI? Which I think this one: https://github.com/UAVCAN/public_regulated_data_types, or do I need to build https://github.com/DroneCAN/libuavcan, (the one printed in the terminal) this module independently?

mertgunduc, 2021-12-15:
I fixed it, here is the solution : [https://github.com/ArduPilot/ardupilot/issues/19523](url)

Use this command to change the url of an existing remote repository:
git remote set-url origin https://github.com/UAVCAN/public_regulated_data_types.git

umer936, 2023-04-19:
Notice all the ones it was unable to fetch start with "git" rather than "https" 

You can use that set-url command above to change each submodule to https, or you can use 

```
git config --global url."https://github.com/".insteadOf git@github.com:
git config --global url."https://".insteadOf git://
```

to set a global config to rewrite the url with "https" instead of "git"