Intelligent-Quads/iq_tutorials #36: Failed to update submodule

i-m-abbhay, 2022-03-15:
I am unable to run "git submodule update --init --recursive"

krum-cck, 2022-03-27:
Hi, I am having the same issue updating submodules. It appears that the cause is Github stopped the usage of URL starting with 'git://'. Did you solve it ?

i-m-abbhay, 2022-03-27:
No not till now. I thought there's some issue with my OS. I have already tried reinstalling Ubuntu on my machine but still getting same error.

MarsJ48, 2022-04-27:
Any updates guys?

i-m-abbhay, 2022-04-27:
No, Still struggling to figure it out... I got busy in my academics due to Endsemester Exams.

French-isotope, 2022-06-23:
Hi @i-m-abbhay @krum-cck @MarsJ48

You could try to use 

`git checkout Copter-4.1.5`

and then : 

`git submodule update --init --recursive` it should be good, let me know if not.

rabah1991, 2022-07-20:
Hi, for me i solved it on Ubuntu 20.04 by using the latest copter version instead of 4.0.4
 It would be as follow:

`git checkout Copter-4.2`

and then :
`
git submodule update --init --recursive`

closed 2023-06-01 by i-m-abbhay