Intelligent-Quads/iq_tutorials #50: [ info] [1691320344.295841232]: waiting for fcu connection

HamidTheCoder, 2023-08-06:
HI there!
I follow this tutorial, but when I run:
rosrun iq_gnc square
I encounter with the following output:
[ info] [1691320344.295841232]: waiting for fcu connection.
How can I solve this issue?
I welcome any feedback.
Thank you.

ARidwanW, 2023-08-21:
Can you give more detail about what did you do and your output? or maybe you can look for references from previous issues related to apm.launch

tomassgomess, 2024-03-18:
This appens to me too!

tomas@tomas-IdeaPad-Gaming:~/catkin_ws/src$ rosrun iq_gnc square
[ INFO] [1710775333.026367211]: using default namespace
[ INFO] [1710775333.034262587]: Waiting for FCU connection

Can anyone help me?
Thanks