Intelligent-Quads/iq_tutorials #55: ERROR: cannot launch node of type [darknet_ros/darknet_ros]:

davitdavtyan, 2024-02-08:
Hi Community,

Has anyone faced with this?

roslaunch darknet_ros darknet_ros.launch 
... logging to /home/eyera/.ros/log/24b60e8c-c6bf-11ee-b2f0-e951cbf0d646/roslaunch-eyera-15148.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://eyera:41993/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /home/eyera/catki...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /webcam/image_raw
 * /darknet_ros/weights_path: /home/eyera/catki...
 * /darknet_ros/yolo_model/config_file/name: yolov2-tiny.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov2-tiny.weights
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

bulentkipcak, 2024-02-09:
you should build the darknet_ros with using this commands:

chmod +x ~/catkin_ws/src/darknet_ros
catkin build -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/gcc-8
catkin build darknet_ros -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/gcc-8
catkin build darknet_ros

davitdavtyan, 2024-02-09:
@bulentkipcak thanks much for replying, so basically "chmod +x ~/catkin_ws/src/darknet_ros" is what were missing in described  steps. And it needs to be in prior of build. 
Actually yesterday I did chmod after buid but that did not help, then I rebuild and that helped, and from what you've written I realized it is because in my second attempt the directory was set to executable. Not sure though why it should be the case.

Thanks again for help.

closed 2024-02-09 by davitdavtyan